Milesight-IoT/SensorDecoders #29: UC51x decoder doesn't correctly handle unknown/new data items

danak6jq, 2023-09-17:
The UC51x decoder is structured with if {} else if {} else if {} else break. We had a older decoder installed on ChirpStack and did not realize it. Newer UC51x v3 has new data items that the decoder does not recognize, which fall down to the break - meaning a single new data item in a payload stops decoding the remainder of the payload.

Suggest removing the 'else' and just test for each known data item. Unknown/new data items (like in UC51x v4) should be ignored without terminating decoding.

milesight-sway, 2024-04-09:
We recommend updating to the latest decode function to ensure the accuracy of the data. Continuing to read data without updating may lead to unforeseen errors.

danak6jq, 2024-04-09:
I understand the latest decoder is a good thing, but this reports an issue where a new mote firmware effectively regresses /breaks an existing decoder unexpectedly. Unknown/new data items (like in UC51x v4) should be ignored without terminating decoding prematurely.

milesight-sway, 2024-04-09:
On the UC50x v3, the upstream link comprises A B C. However, on UC50x v4, it includes an additional element, "D", which the decoder for the older version cannot recognize. Continuing decoding might result in misinterpretation or other implications. 

Since the decoder function doesn't have memory capabilities, we can only proceed by simply disregarding it.

If you have better suggestions, we're open to optimization.

danak6jq, 2024-04-10:
On the UC50x v4, is the uplink always in the sequence A B C D, or can D be earlier in the uplink? I've observed that case, for example,  A B D C. The way the decoder is structured, C will not be decoded at all by the v3 decoder because the decoding loop 'breaks' at an unknown element.  By eliminating the break, the decoder will simply ignore D, and decode C as expected. 

While I agree decoders should be kept up to date, this isn't always the case operationally. A simple change - eliminating 'else break' - maintains compatibility until the decoder is updated

milesight-sway, 2024-04-11:
I understand your requirements well. The sequence A B D C is just a simple example. However, within D, it could potentially contain the starting point of any fragment, leading to complete data disarray. This flaw is inherent in our protocol design.
```
Milesight Device Protocol Defination

+--------------------------------------+
|                  A                   |
+------------+-----------+-------------+
| CHANNEL_ID | DATA_TYPE |     DATA    |
+------------+-----------+-------------+

+---------------+---------------+---------------+---------------+
|       A       |       B       |       D       |       C       |
+---------------+---------------+---------------+---------------+
|  01 0A 00 00  |  02 0B 00 00  |  04 0D 03 0C  |  03 0C 00 00  |
                                         -----     ----- 
                   DECODE BEGINING:     (ERROR)   (ACTUAL)

```

closed 2024-05-09 by Milesight-IoT