Milesight-IoT/SensorDecoders #33: AT101 payload and decoder incorrectly uses positive longitude for West

danak6jq, 2023-11-26:
I'm located in California, near 38.169N and 122.137W. By convention, southern latitudes and western longitudes are given a negative sign, so this should be:

38.169, -122.137

However, the AT101 payload and decoded value shows 38.169, 122.137.

If I give this string to Google Maps, it shows the correct location in California:

[image]

However, the AT101 payload and decoded value shows 38.169, 122.137, which is in the sea between China and Korea:

[image]

Perhaps the decoder should multiply the longitude by -1 ?

milesight-sway, 2023-11-28:
what's the raw payload data?

danak6jq, 2023-11-28:
Raw payload in hex here:
[image]

Decoded payload here, along with raw payload in base64
[image]

The Chirpstack v4 decoder:
[image]

Source for the decoder:
https://github.com/Milesight-IoT/SensorDecoders/blob/main/AT_Series/AT101/AT101_Chirpstack_v4.js

You can manually decode the payload like I did and confirm I'm seeing a positive longitude value in the AT101
payload; the decoder is just converting that correctly.

danak6jq, 2023-11-28:
> 1. What's the raw data?
> 2. What decoder are you using?

Could you share raw payloads from China? Perhaps the longitude completely lacks E/W sign?

milesight-sway, 2023-11-28:
What is the firmware version of the AT101?

ToolBox App
- iOS (https://apps.apple.com/us/app/milesight-toolbox/id1518748039)
- Android (https://play.google.com/store/search?q=milesight+toolbox&c=apps)

milesight-sway, 2023-11-28:
Upgrade AT101 firmware using ToolBox App. 
[AT101.0000.0100.0101-r1.zip](https://github.com/Milesight-IoT/SensorDecoders/files/13484231/AT101.0000.0100.0101-r1.zip)

danak6jq, 2023-11-28:
> What is the firmware version of the AT101?
> 
> ToolBox App
> 
> * iOS (https://apps.apple.com/us/app/milesight-toolbox/id1518748039)
> * Android (https://play.google.com/store/search?q=milesight+toolbox&c=apps)

[image]

danak6jq, 2023-11-28:
> Upgrade AT101 firmware using ToolBox App. （rename .zip -> .bin） [AT101.0000.0100.0101-r1.zip](https://github.com/Milesight-IoT/SensorDecoders/files/13484231/AT101.0000.0100.0101-r1.zip)

Upgraded firmware, here's the first report - seems to have addressed this issue. Will this firmware be released normally?

Thank you!

[image]

danak6jq, 2023-11-28:
I do see battery level of 1%, which might be my fault (I had difficulty upgrading the firmware over NFC, seemed to brick the AT101, removed the batteries, was eventually successful with BLE).

It's possible the battery report will correct itself?

milesight-sway, 2023-11-29:
The firmware file is a beta version.

Regarding the battery problem, try removing the battery for a while and then reinstalling it and expect that to solve your problem!

danak6jq, 2023-11-29:
> The firmware file is a beta version.
> 
> Regarding the battery problem, try removing the battery for a while and then reinstalling it and expect that to solve your problem!

That resolved the issue - 30 minutes no batteries, I now get correct reports.

Any idea when this beta firmware build will be in release?

Thank you!

milesight-sway, 2023-11-30:
Yes.

We appreciate your feedback to help us make our product even better.

closed 2023-12-07 by Milesight-IoT