Milesight-IoT/SensorDecoders #38: Why are there separate Chirpstack (v3) and ChirpStack v4 decoders? Single decoder works

danak6jq, 2024-04-06:
There's no need for two different decoders, especially since this creates the chance of error where the actual payload decoding does not match.

Chirpstack v3 calls this function:

```
// Decode decodes an array of bytes into an object.
//  - fPort contains the LoRaWAN fPort number
//  - bytes is an array of bytes, e.g. [225, 230, 255, 0]
//  - variables contains the device variables e.g. {"calibration": "3.5"} (both the key / value are of type string)
// The function must return an object, e.g. {"temperature": 22.5}
function Decode(fPort, bytes, variables) {
  return {};
}
```

Chirpstack v4 calls this function:

```
// Decode uplink function.
//
// Input is an object with the following fields:
// - bytes = Byte array containing the uplink payload, e.g. [255, 230, 255, 0]
// - fPort = Uplink fPort.
// - variables = Object containing the configured device variables.
//
// Output must be an object with the following fields:
// - data = Object representing the decoded payload.
function decodeUplink(input) {
  return {
    data: {
      temp: 22.5
    }
  };
}
```

A single decoder file may contain both functions, which call a single actual decoding code.
Thus a single file will serve for both v3 and v4, allowing MileSight to make changes to a single file and users to use a single file, preventing errors.

milesight-sway, 2024-04-09:
That's a good idea, thanks for the offer!

danak6jq, 2024-04-09:
I'll do a PR today  with a suggested change

milesight-sway, 2024-04-09:
Thank you for your participation, it has been updated as per your suggestion.

closed 2024-05-09 by Milesight-IoT