Milesight-IoT/SensorDecoders #41: TS30X Payload Definition doesnt fit with decoder

NicciZar, 2024-06-05:
Hi,

Payload definition of TS30x doesn't say that "magnet_chn1_alarm" exists but in the decoder.js it does exist.
You'll find it on line 109

```JS               case 0x05:
line 108                    data.magnet_chn1 = readInt16LE(bytes.slice(i, i + 2)) === 0 ? "closed" : "opened";
line 109                    data.magnet_chn1_alarm = "threshold";
line 110                    break;
```
Is this correct or should this be a different property?

milesight-sway, 2024-06-06:
https://github.com/Milesight-IoT/SensorDecoders/tree/main/TS_Series/TS30x#historical-data-channel-mask

milesight-sway, 2024-07-01:
Since we haven't heard back from you for a while, we will proceed to close this issue. If you encounter any other problems, don't hesitate to open an issue.

closed 2024-07-09 by Milesight-IoT