Milesight-IoT/SensorDecoders #42: get devEUI inside the decoder

salembream, 2024-06-17:
Hello,

I'm testing the UG56 GW, to get data from em300-DI sensors, with custom decoder (copied from this github repo).
My system/application receive the data through the MQTT interface in the gateway.

the issue is that the received MQTT message is the same as the decoder output which does not include the devEUI.
so my question, is there a way to read/know the devEUI within the decoder, so we can add it to the output message?

mscloud-1, 2024-06-18:
READ: https://github.com/Milesight-IoT/SensorDecoders/blob/main/LoRaObject.md
- `LoRaObject.devEUI`

milesight-sway, 2024-06-18:
READ: https://github.com/Milesight-IoT/SensorDecoders/blob/main/LoRaObject.md
- `LoRaObject.devEUI`

closed 2024-06-18 by salembream