Milesight-IoT/SensorDecoders #43: Problem decoding up messages in EM300DI

isabido, 2024-06-20:
HI, we are integrating this EM300-DI device into Chirpstack 4.8.1, after loading the decode that you have published.

https://github.com/Milesight-IoT/SensorDecoders/blob/main/EM_Series/EM300_Series/EM300-DI/EM300-DI_Decoder.js

I have a configuration where the totalizer of a water meter is sent every minute.

And I get these error messages randomly due to decoding failure.
[image]

Here is an image of a UP that cannot decode
[image]

And here is an image of a UP that decodes correctly
[image]
[EM-300-DI.log.json](https://github.com/user-attachments/files/15920452/EM-300-DI.log.json)

I am attaching a more complete log where you can see UP messages correctly decoded and others that are not.

What could be happening?

thanks for your help

danak6jq, 2024-06-20:
@isabido could you please copy and paste the _text_ of the 'data:' item from the two UPs? I'd rather not type them in and perhaps have an error.

This:
[image]

and this:
[image]

Thank you -
Dana

isabido, 2024-06-21:
Hi Dana, 

Thank you for your help, sorry I sent the image of the same package, it was a mistake, I'll give you this new example that is about the same thing.

[image]

This would be the one that returns the error 
data:"AXVkA2cYAQRoZQXhCgAKAAA0skY="
[image]

And this one that sends the next minute would be correct
data:"AXVkA2cZAQRoZAXhCgAKAAA0skY="
[image]

milesight-sway, 2024-07-01:
It appears you have already reached out to our technical support team. We will go ahead and close this issue.

The decoding function for the EM300-DI is working without any problems.

danak6jq, 2024-07-03:
Just to complete my comments - I manually decoded the "good" and "bad" payloads and they're both good, in fact nearly identical. Glad you're getting it resolved.

closed 2024-07-09 by Milesight-IoT