Milesight-IoT/SensorDecoders #44: UC50x decoder doesn't work in ChripStack v4

Sleeper67, 2024-06-24:
Hi there,

I tried to upload the codec from UC_Series/UC50x/UC50x_Decoder.js and it doesn't work instead it logs me out of ChirpStack.

Ended up having to clean the script to allow it to save into ChirpStack v4.

milesight-sway, 2024-06-26:
I tried using [uc50x_decoder.js](https://github.com/Milesight-IoT/SensorDecoders/blob/main/UC_Series/UC50x/UC50x_Decoder.js) in ChirpStack v4, and it didn't cause any errors. Could you provide more details so I can assist you better?

- chirpstack v4 (Payload codec -- JavaScript functions -- codec functoins)

Sleeper67, 2024-06-29:
Hi there,

I'm not it's a problem sorry, after carrying out more testing.

I have found a bug that seems to be with my sub-domain name to my ChirpStack instance not allowing a POST and then logging me out every time I try to update the codec.

It was just unusual behaviour, that I have not come across,  until I tired to use this script.

Thanks.

milesight-sway, 2024-07-01:
Great to hear back from you. If you encounter any other problems, don't hesitate to open an issue.

closed 2024-07-09 by Milesight-IoT