Milesight-IoT/SensorDecoders #45: Set via Lorawan

corwin1807, 2024-07-02:
Good afternoon, 

Could you tell me, is it possible to manage Milesight devices via Lorawan downlink? 
For example Set Transmit Interval Time, Restart, Turn On/Off, rejoin interval

Thanks for any information.

milesight-sway, 2024-07-03:
Milesight devices support sending configuration commands, such as setting reporting intervals, collection intervals, rebooting the device, and more. 

Such as AM300 Series：
[Milesight Website](https://www.milesight.com/iot/resources/download-center/) -> Technical Documents -> AM Series -> AM300 & AM300L Series -> AM300(L) Series User Guide

The documentation contains information about Payloads. （Uplink / Downlink）

milesight-sway, 2024-07-09:
Since we haven't heard back from you for a while, we will proceed to close this issue. If you encounter any other problems, don't hesitate to open an issue.

closed 2024-07-09 by Milesight-IoT