Milesight-IoT/SensorDecoders #50: No Encoders for EM310-UDL or WS201 or WS202 or EM300-SLD.

Sleeper67, 2024-08-12:
Hi there,

Will you be releasing any other encoders?

There are No Encoders for EM310-UDL or WS201 or WS202 or EM300-SLD.

Thanks

milesight-sway, 2024-08-13:
Payload Encoder (Preview Version)

- EM310-UDL: https://github.com/Milesight-IoT/SensorDecoders/blob/feat/em310-udl/EM_Series/EM300_Series/EM310-UDL/EM310-UDL_Encoder.js
- WS201: https://github.com/Milesight-IoT/SensorDecoders/blob/feat/ws201/WS_Series/WS201/WS201_Encoder.js
- WS202: https://github.com/Milesight-IoT/SensorDecoders/blob/feat/ws202/WS_Series/WS202/WS202_Encoder.js
- EM300-SLD: https://github.com/Milesight-IoT/SensorDecoders/blob/feat/em300-sld/EM_Series/EM300_Series/EM300-SLD/EM300-SLD_Encoder.js

Sleeper67, 2024-08-15:
Thank you  @milesight-sway

closed 2024-08-15 by Sleeper67