Milesight-IoT/SensorDecoders #51: Missing Definition of pwoer factor for ws523

theArcher73, 2024-08-13:
Hi,

we use the WS523 and see the information power factor into the payload. It's a percentage but we cant find any information a percentage from what? a percentage of the max. currency (16A) or a percentage for the current value of the overload protection?

I have not found anything about this in the data sheet, the instructions or here in the GIT. 

And btw. there is no picture für the socket to use in germany and other countries with similar plugs in the git ;-)

With best regards,
S.

milesight-sway, 2024-08-13:
For more information about our products, please refer to the details below：
- [Milesight IoT Products - WS52x](https://www.milesight.com/iot/product/lorawan-sensor/ws523)

milesight-sway, 2024-08-14:
An example showing that `087001 05812C 07C94A00 03743009 068309660000 048007000000` (`048007000000`) represents `active_power = 7w` has been included.
- `power_factor = P / S` （P: Active Power, S: Apparent Power）
The power factor data, originating from the chip, inherently has a certain margin of error.

The [README](https://github.com/Milesight-IoT/SensorDecoders/tree/main/WS_Series/WS52x) of WS52x has been updated with product images.
- Support Socket & Plug Type：US / EU / AU / UK / CN
If you require sockets of different standards, please visit [official website](https://www.milesight.com/company/contactus) to consult with our sales team.

theArcher73, 2024-08-14:
Hi,

ok, I saw that "active power" is in the payload but **not in your payload-decoder** you provided
power_factor = P / S （P: Active Power, S: Apparent Power） - ok, that helps. 

Thx,

S.

milesight-sway, 2024-08-14:
I sincerely apologize for any inconvenience this may have caused and greatly appreciate your feedback, which helps us enhance our product. In response to your suggestions, we have now added detailed product images and refined the previously unclear payload-decoder. 

Thank you very much. : )

theArcher73, 2024-08-19:
Is there a range for the WS523 within which the delay for delayed tasks can be selected? (sorry, right now at this point to define the boundaries for the downlink parameter?)

theArcher73, 2024-08-19:
ok, i searched it in the manual eg the documentation. I overlooked the encoder completely - but it helps!
Thx

theArcher73, 2024-09-05:
Hello Sway,

sorry, but in the decoder I already missed the part for the Power Outage Alert. I have an idea but be not shure. Thats why I reopen that issue in the hope that than all things are done.

When there are missing informations into the datasheet - where is the right place to ask for this informations?

I mean:

- min/max. deviation of the measured values (
- Accuracy of the measured values (Number of decimal places of values)
- min/max. range of the individual measured values (e.g. Voltage minimum 0V, max. 250V but can read this sensor also supply and feed-in when a solar panel is plugged in?)

Protection class of the housing (I found IP30 in a webshop), but this is not in the data sheet

Steffen

Milesight-IoT, 2024-09-09:
Please Contact Support Team for More Help. (This repo is only for Sensor Codec issues)

1. Go to https://support.milesight-iot.com/support/home
2. create a new ticket
3. balabala

closed 2024-12-26 by milesight-sway