Milesight-IoT/SensorDecoders #52: CT101  uplink stop after report interval from 1 hour to 5 min

Sleeper67, 2024-09-04:
Hi there,

I changed my device (CT101) Uplink interval from 3600s to 300s via downlink and now the device has stopped sending uplinks.

Any idea why it would stopped?

Had the uplinks also recently stop on one of our EM320-TH-915M, which was sending confirmed uplinks 

we're running Chirpstack v4.9.0

[image]

Milesight-IoT, 2024-09-04:
According to your screenshot, you have set `FF 8E 01 2C 01`, which means that the data will be reported after 300 minutes.
- [CT101 Downlink setReportInterval](https://github.com/Milesight-IoT/SensorDecoders/blob/1757419059092f333db217d340d677850cc73d4c/CT_Series/CT101/CT101_Encoder.js#L71)

Milesight-IoT, 2024-09-04:
Due to lack of other context, it is not possible to determine why EM320-TH has stopped transmitting. You can check the device's network status, reporting interval, and whether there is sufficient power using Milesight ToolBox.

Milesight ToolBox :
- [Apple Store](https://apps.apple.com/us/app/milesight-toolbox/id1518748039) 
- [Google Play](https://play.google.com/store/search?q=milesight+toolbox&c=apps&hl=en)

Sleeper67, 2024-09-04:
> According to your screenshot, you have set `FF 8E 01 2C 01`, which means that the data will be reported after 300 minutes.
> 
> * [CT101 Downlink setReportInterval](https://github.com/Milesight-IoT/SensorDecoders/blob/1757419059092f333db217d340d677850cc73d4c/CT_Series/CT101/CT101_Encoder.js#L71)

Thanks for the reply, I'll change my encoder code to suit for consistency

My assumption was that the encoder would stay consistent and use the same units for setting the reporting interval. Since the encoder for the EM320-TH uses seconds, I expected the encoder for the CT101 to use seconds as well when configuring the report_interval.

Sleeper67, 2024-09-04:
> Due to lack of other context, it is not possible to determine why EM320-TH has stopped transmitting. You can check the device's network status, reporting interval, and whether there is sufficient [...]

The EM320-TH has stopped sending join request as far as I can tell, unfortunately the device is far from us.

[image]

closed 2024-12-23 by Sleeper67