Milesight-IoT/SensorDecoders #56: QUESTION - Why I don't get temperature inside a pyload from my Milesight AT101 GPS Tracker

IhorYednak, 2024-12-09:
Hello,

Recently I added my Milesight AT101 GPS Tracker to the thingsboard instance through the KPN things integration, but I don't get any payload with a temperature values.
Here are all payloads I got from the device:

```
{
    "deviceName": "urn:dev:DEVEUI:24E124745E111889:",
    "deviceLabel": "Milesight AT101 GPS Tracker",
    "deviceType": "Own LoRa device (preset connectivity)",
    "telemetry": [{
        "ts": 1733735427753,
        "values": {
            "payload": "06d9aa0c8e294fc014a20306d9aa64cc226e10efa703",
            "port": 85,
            "timeOrigin": "NETWORK"
        }
    }]
}
===========================================================================

{
    "deviceName": "urn:dev:DEVEUI:24E124745E111889:",
    "deviceLabel": "Milesight AT101 GPS Tracker",
    "deviceType": "Own LoRa device (preset connectivity)",
    "telemetry": [{
        "ts": 1733735427753,
        "values": {
            "payload": "06d9aa0c8e294fc014a20306d9aa64cc226e10efa703",
            "port": 85,
            "FCntUp": 7159,
            "FCntDn": 169,
            "LrrRSSI": -112,
            "LrrSNR": -9,
            "LrrESP": -121.514969,
            "DevLrrCnt": 2,
            "SpFact": 10,
            "SubBand": "G2",
            "Channel": "LC11",
            "timeOrigin": "NETWORK"
        }
    }]
}
===========================================================================

{
    "deviceName": "urn:dev:DEVEUI:24E124745E111889:",
    "deviceLabel": "Milesight AT101 GPS Tracker",
    "deviceType": "Own LoRa device (preset connectivity)",
    "telemetry": [{
        "ts": 1733735419249,
        "values": {
            "locOrigin": "KPNLORA",
            "latitude": 51.953934,
            "longitude": 5.23532,
            "radius": 212.319489,
            "locMethod": "TDOA",
            "locAccuracy": 9999,
            "locPrecision": 9999,
            "locTime": 1.733735419249E9
        }
    }]
}
===========================================================================

{
    "deviceName": "urn:dev:DEVEUI:24E124745E111889:",
    "deviceLabel": "Milesight AT101 GPS Tracker",
    "deviceType": "Own LoRa device (preset connectivity)",
    "telemetry": [{
        "ts": 1733735418917,
        "values": {
            "payload": "01755a0367560005000006d9aa78968272304da40306d9aa082697d58ae1ac0306d9aa0a2697d58ae1ab03",
            "port": 85,
            "timeOrigin": "NETWORK"
        }
    }]
}
===========================================================================

```
Also here is an uplink decoder i use to get the data out of the integration:

```
// Function to decode payload (array of numbers) to a string
function decodePayload(payload) {
    return String.fromCharCode.apply(null, payload);
}

// Decode payload to string
var payloadStr = decodePayload(payload);

// Initialize telemetryData array
var telemetryData = [];
var debugMessages = [];

// Check if metadata is defined
if (metadata) {
    // Extract deviceName and deviceType from the payload
    var deviceLabel = "unknown";
    var deviceType = "unknown";

    // If metadata.integrationName is provided, use it
    if (metadata.integrationName) {
        deviceLabel = "integration_" + metadata.integrationName;
        deviceType = "integration_device";
    }

    // Parse the payload as JSON
    try {
        var jsonPayload = JSON.parse(payloadStr);

        // Initialize attributes object
        var attributes = {
            integrationName: metadata.integrationName || "unknown"
        };

        // Check if latitude and longitude are present in the telemetry values
        if (jsonPayload.telemetry && jsonPayload.telemetry.length > 0) {
            jsonPayload.telemetry.forEach(function(entry) {
                if (entry.values.latitude !== undefined && entry.values.longitude !== undefined) {
                    debugMessages.push("Latitude and Longitude found: " + entry.values.latitude + ", " + entry.values.longitude);
                    attributes.latitude = entry.values.latitude;
                    attributes.longitude = entry.values.longitude;
                } else {
                    debugMessages.push("Latitude and Longitude are not present in the telemetry values.");
                }

                // Add telemetry data (excluding latitude and longitude)
                telemetryData.push({
                    ts: entry.ts,
                    values: entry.values
                });
            });
        }

        // Build result object
        var result = {
            deviceName: jsonPayload.deviceLabel || deviceLabel,
            deviceType: jsonPayload.deviceType || deviceType,
            attributes: attributes,  // Add the modified attributes here
            telemetry: telemetryData,
            debug: debugMessages // Include debug messages for tracing
        };

        return result;  // Return the full result object, including latitude and longitude as attributes
    } catch (e) {
        // Handle JSON parse errors
        return {
            deviceLabel: deviceLabel,
            deviceType: deviceType,
            attributes: {
                integrationName: metadata.integrationName || "unknown"
            },
            telemetry: [],
            debug: ["JSON parse error: " + e.message]
        };
    }
} else {
    // If metadata is completely missing
    return {
        deviceName: "unknown",
        deviceType: "unknown",
        attributes: {
            integrationName: "unknown"
        },
        telemetry: [],
        debug: ["Metadata is undefined."]
    };
}

```

Could you please advise or maybe there are simply no that kind of a data, because based on the user guide - there are a lot of interesting things including this temperature.
Thank you in advance!

Milesight-Lockon, 2024-12-10:
Hi,
Based on your description, there are two payload examples in your content:
1.  _06d9aa0c8e294fc014a20306d9aa64cc226e10efa703_
2.  _01755a0367560005000006d9aa78968272304da40306d9aa082697d58ae1ac0306d9aa0a2697d58ae1ab03_

It seems that you have enabled the WiFi positioning function.
In the two examples above, the payload starting with **06d9** is specifically used for "**WiFi scan result reporting**" so it does not contain a temperature field.
And the payload starting with **0175** is the normal reporting payload of the device, which includes the temperature field. 

If you are particularly concerned about the temperature field, I suggest extracting it from the payloads starting with **0175**.

Special Note:
You can refer to section 5.1 "Example" of the [AT101 User Guide](https://resource.milesight.com/milesight/iot/document/at101-user-guide-en.pdf) for more details.

IhorYednak, 2024-12-19:
Thank you for your answer @Milesight-Lockon, looks good now :)

closed 2024-12-19 by IhorYednak