Milesight-IoT/SensorDecoders #59: Encoders for the UC300 and UC50X

Sleeper67, 2024-12-28:
Hi @milesight-sway ,

Could I also please get Encoders for the UC300 and UC50X.

Thanks

FengFuLiu, 2024-12-31:
Do you mean [UC300]( https://github.com/Milesight-IoT/SensorDecoders/tree/main/UC_Series/UC300) &  [UC50X](https://github.com/Milesight-IoT/SensorDecoders/tree/main/UC_Series/UC50x)

Sleeper67, 2025-01-08:
Hi, was there an encoder i can get for these 2 x controllers?

Thanks

milesight-sway, 2025-01-15:
beta version
- [uc300_encoder](https://github.com/Milesight-IoT/SensorDecoders/blob/feat/uc300/UC_Series/UC300/UC300_Encoder.js)
- [uc50x_encoder](https://github.com/Milesight-IoT/SensorDecoders/blob/feat/uc50x/UC_Series/UC50x/UC50x_Encoder.js)

Sleeper67, 2025-01-15:
Thank you @milesight-sway

closed 2025-01-15 by Sleeper67