Milesight-IoT/SensorDecoders #62: EM300-MCS/EM300-SLD battery payload missing

peternmuller, 2025-01-27:
When getting payload data from EM300-MCS/EM300-SLD sensors, there is no battery information. I saw that in the code, there is a function to decode battery, though.
I tried with the correct payload decoder for each sensors, but still no data.

Here's what I get for the EM300-MCS:
```
temperature: 21.9
humidity: 53.5
magnet_status: "close"
```

For the EM300-SLD:
```
temperature: 21.1
humidity: 54.5
leakage_status: "normal"
```

I also tested with the EM320-TH sensor, and this time it reports the battery as expected:
```
battery: 100
temperature: 23.8
humidity: 39
```

I also tried walking through the MIB, and there is no battery for EM300-MCS/EM300-SLD sensors:
```
iso.3.6.1.2.1.47.1.1.1.1.7.2200 = STRING: "2001 Temperature"
iso.3.6.1.2.1.47.1.1.1.1.7.2201 = STRING: "2001 Humidity"
iso.3.6.1.2.1.47.1.1.1.1.7.2202 = STRING: "2001 Leak"
iso.3.6.1.2.1.47.1.1.1.1.7.2300 = STRING: "1745 Temperature"
iso.3.6.1.2.1.47.1.1.1.1.7.2301 = STRING: "1745 Humidity"
iso.3.6.1.2.1.47.1.1.1.1.7.2304 = STRING: "1745 Magnet"
```

peternmuller, 2025-01-27:
Didn't touch it for about an hour, the EM300-MCS started reporting battery stats, but the issue with EM300-SLD still persists.

nolimitdev, 2025-02-26:
We have several EM300-SLD with diff HW and FW versions and they are sending battery, but very rarely. We have 5 min reporting interval and battery is sent in about 3 times per day.  Log all data and battery must be there. Also check device options. Im not sure if battery reporting can be disabled. Maybe you disabled it by mistake.

milesight-sway, 2025-02-27:
The battery of a LoRa device typically lasts for more than 5 years, with daily battery drain being minimal. However, since each data transmission consumes some power, we have implemented several strategies to further maximize the battery's lifespan.

Different firmware may have different battery report intervals (e.g., 6 hours per day).

peternmuller, 2025-02-28:
First of all, thanks for the replies. I also noticed that after a while, the battery statistics appear. In fact, they take much longer to appear on the EM300-SLD/MCS than on the EM320-TH sensors, where the battery statistics are reported every time.

closed 2025-02-28 by peternmuller

milesight-sway, 2025-03-05:
influenced by the device's detection mechanism, whether it is active monitoring or passive triggering.
- EM300-SLD / EM300-MCS (passive trigger)
- EM300-TH (active monitor)