Milesight-IoT/SensorDecoders #67: Decoder for WTS506

ivanzy, 2025-03-26:
The decoder for the WTS506 device is not parsing rainfall correctly. The problem is in the channel_type conditional of the decoder that should `0xEC` instead of `0x77.

milesight-sway, 2025-03-27:
- use this version [WTS(v3) ](https://github.com/Milesight-IoT/SensorDecoders/blob/211df844e78d4404302ba4a3feb9f7aee26e7732/WTS_series/WTS_Decoder.js#L68)

``` javascript
// RAINFALL TOTAL
else if (channel_id === 0x08 && channel_type === 0x77) {
    decoded.rainfall_total = readUInt16LE(bytes.slice(i, i + 2)) / 100;
    decoded.rainfall_counter = bytes[i + 2];
    i += 3;
}
// RAINFALL TOTAL (v3)
else if (channel_id === 0x08 && channel_type === 0xec) {
    decoded.rainfall_total = readUInt32LE(bytes.slice(i, i + 4)) / 100;
    decoded.rainfall_counter = bytes[i + 4];
    i += 5;
}
```

milesight-sway, 2025-04-03:
Due to a lack of feedback for an extended period, we will close this issue. If you have any further questions, please feel free to open a new issue.

closed 2025-04-03 by milesight-sway