Milesight-IoT/SensorDecoders #69: EM300-DI Payload Codec Pulse

Keelan, 2025-04-04:
Shouldn't the below be pulse not water?  If i've configured the sensor as counter/pulse my pulse value is always 0, but i see the water go up.

https://github.com/Milesight-IoT/SensorDecoders/blob/518cd81a223ffc728dc097b57f8a9162e8141d76/EM_Series/EM300_Series/EM300-DI/EM300-DI_Decoder.js#L65

What i changed my codec too:
`decoded.pulse = readFloatLE(bytes.slice(i + 4, i + 8)); `

Also in the pulse context do we need the following? 
`decoded.water_conv = readUInt16LE(bytes.slice(i, i + 2)) / 10;` 

We already have
`decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;`

Keelan, 2025-04-04:
Shouldn't we also be writing 1 or 2 as gpio_type not 0 and 1?
So the MSV work on BACnet. (MSVs start at 1 not 0)

Current
```
        // GPIO
        else if (channel_id === 0x05 && channel_type === 0x00) {
            decoded.gpio = bytes[i];
            decoded.gpio_type = 0;
            i += 1;
        }
        // PULSE COUNTER
        else if (channel_id === 0x05 && channel_type === 0xc8) {
            decoded.pulse = readUInt32LE(bytes.slice(i, i + 4));
            decoded.gpio_type = 1;
            i += 4;
        }
        // PULSE COUNTER (v1.3+)
        else if (channel_id === 0x05 && channel_type === 0xe1) {
            decoded.water_conv = readUInt16LE(bytes.slice(i, i + 2)) / 10;
            decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;
            decoded.water = readFloatLE(bytes.slice(i + 4, i + 8));
            decoded.gpio_type = 1;
            i += 8;
        }
```

Corrected
```
        // GPIO
        else if (channel_id === 0x05 && channel_type === 0x00) {
            decoded.gpio = bytes[i];
            decoded.gpio_type = 1;
            i += 1;
        }
        // PULSE COUNTER
        else if (channel_id === 0x05 && channel_type === 0xc8) {
            decoded.pulse = readUInt32LE(bytes.slice(i, i + 4));
            decoded.gpio_type = 2;
            i += 4;
        }
        // PULSE COUNTER (v1.3+)
        else if (channel_id === 0x05 && channel_type === 0xe1) {
            decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;
            decoded.pulse = readFloatLE(bytes.slice(i + 4, i + 8));
            decoded.gpio_type = 2;
            i += 8;
        }
```

Keelan, 2025-04-04:
Also in history v2

Current
```
            if (mode === 1) {
                data.gpio = bytes[i + 9];
            } else if (mode === 2) {
                data.water_conv = readUInt16LE(bytes.slice(i + 10, i + 12)) / 10;
                data.pulse_conv = readUInt16LE(bytes.slice(i + 12, i + 14)) / 10;
                data.water = readFloatLE(bytes.slice(i + 14, i + 18));
            }
```

Adjusted
```
            if (mode === 1) {
                data.gpio = bytes[i + 9];
            } else if (mode === 2) {
                data.water_conv = readUInt16LE(bytes.slice(i + 10, i + 12)) / 10;
                data.pulse_conv = readUInt16LE(bytes.slice(i + 12, i + 14)) / 10;
                data.water = readFloatLE(bytes.slice(i + 14, i + 18));
                **data.pulse = readFloatLE(bytes.slice(i + 14, i + 18));**
            }
```

Charles-Milesight, 2025-04-12:
> Shouldn't the below be pulse not water? If i've configured the sensor as counter/pulse but in the payload JSON the pulse is missing and in bacnet the pulse value is always 0, but i see the water go up. The pulses i am receiving to the sensor are gas and electrical not water. So makes sense its called _pulse_ rather than _water_
> 
> [SensorDecoders/EM_Series/EM300_Series/EM300-DI/EM300-DI_Decoder.js](https://github.com/Milesight-IoT/SensorDecoders/blob/518cd81a223ffc728dc097b57f8a9162e8141d76/EM_Series/EM300_Series/EM300-DI/EM300-DI_Decoder.js#L65)
> 
> Line 65 in [518cd81](/Milesight-IoT/SensorDecoders/commit/518cd81a223ffc728dc097b57f8a9162e8141d76)
> 
>  decoded.water = readFloatLE(bytes.slice(i + 4, i + 8)); 
> What i changed my codec too: `decoded.pulse = readFloatLE(bytes.slice(i + 4, i + 8)); `
> 
> Also in the pulse context do we need the following? `decoded.water_conv = readUInt16LE(bytes.slice(i, i + 2)) / 10;`
> 
> We already have `decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;`
> 
> Similarly, do we need _pulse_conv_ in the _water alarm_ section?
> 
> **from**
> 
> ```
>         // WATER ALARM
>         else if (channel_id === 0x85 && channel_type === 0xe1) {
>             decoded.water_conv = readUInt16LE(bytes.slice(i, i + 2)) / 10;
>             decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;
>             decoded.water = readFloatLE(bytes.slice(i + 4, i + 8));
>             decoded.water_alarm = readWaterAlarm(bytes[i + 8]);
>             i += 9;
>         }
> ```
> 
> **to**
> 
> ```
>         // WATER ALARM
>         else if (channel_id === 0x85 && channel_type === 0xe1) {
>             decoded.water_conv = readUInt16LE(bytes.slice(i, i + 2)) / 10;
>             decoded.water = readFloatLE(bytes.slice(i + 4, i + 8));
>             decoded.water_alarm = readWaterAlarm(bytes[i + 8]);
>             i += 9;
>         }
> ```

> Shouldn't the below be pulse not water? If i've configured the sensor as counter/pulse but in the payload JSON the pulse is missing and in bacnet the pulse value is always 0, but i see the water go up. The pulses i am receiving to the sensor are gas and electrical not water. So makes sense its called _pulse_ rather than _water_
> 
> [SensorDecoders/EM_Series/EM300_Series/EM300-DI/EM300-DI_Decoder.js](https://github.com/Milesight-IoT/SensorDecoders/blob/518cd81a223ffc728dc097b57f8a9162e8141d76/EM_Series/EM300_Series/EM300-DI/EM300-DI_Decoder.js#L65)
> 
> Line 65 in [518cd81](/Milesight-IoT/SensorDecoders/commit/518cd81a223ffc728dc097b57f8a9162e8141d76)
> 
>  decoded.water = readFloatLE(bytes.slice(i + 4, i + 8)); 
> What i changed my codec too: `decoded.pulse = readFloatLE(bytes.slice(i + 4, i + 8)); `
> 
> Also in the pulse context do we need the following? `decoded.water_conv = readUInt16LE(bytes.slice(i, i + 2)) / 10;`
> 
> We already have `decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;`
> 
> Similarly, do we need _pulse_conv_ in the _water alarm_ section?
> 
> **from**
> 
> ```
>         // WATER ALARM
>         else if (channel_id === 0x85 && channel_type === 0xe1) {
>             decoded.water_conv = readUInt16LE(bytes.slice(i, i + 2)) / 10;
>             decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;
>             decoded.water = readFloatLE(bytes.slice(i + 4, i + 8));
>             decoded.water_alarm = readWaterAlarm(bytes[i + 8]);
>             i += 9;
>         }
> ```
> 
> **to**
> 
> ```
>         // WATER ALARM
>         else if (channel_id === 0x85 && channel_type === 0xe1) {
>             decoded.water_conv = readUInt16LE(bytes.slice(i, i + 2)) / 10;
>             decoded.water = readFloatLE(bytes.slice(i + 4, i + 8));
>             decoded.water_alarm = readWaterAlarm(bytes[i + 8]);
>             i += 9;
>         }
> ```

For Milesight EM300-DI sensors with firmware version v1.2, the sensor data includes the pulse count. However, starting from firmware version v1.3, the EM300-DI does not report the pulse count. Instead, it reports data such as water_conv, pulse_conv, and water. The calculation formula for water is: water = (pulse / pulse_conv) * water_conv. Therefore, you can add the pulse variable in the related decoder. The formula for calculating pulse is: pulse = (water / water_conv) * pulse_conv. So, if you want to obtain the pulse count in EM300-DI sensors with firmware version 1.3 and above, you can use this decoder(https://resource.milesight-iot.com/Support/Charles/decoder%20for%20EM300-DI.doc) modified by me.

water_conv and pulse_conv are two conversion variables, which indicate how many pulses measure how much water. You can remove either conversion variable according to your needs.

Charles-Milesight, 2025-04-12:
> Shouldn't we also be writing 1 or 2 as gpio_type not 0 and 1? So the MSV work on BACnet. (MSVs start at 1 not 0) and also the history mode selection works?
> 
> Current
> 
> ```
>         // GPIO
>         else if (channel_id === 0x05 && channel_type === 0x00) {
>             decoded.gpio = bytes[i];
>             decoded.gpio_type = 0;
>             i += 1;
>         }
>         // PULSE COUNTER
>         else if (channel_id === 0x05 && channel_type === 0xc8) {
>             decoded.pulse = readUInt32LE(bytes.slice(i, i + 4));
>             decoded.gpio_type = 1;
>             i += 4;
>         }
>         // PULSE COUNTER (v1.3+)
>         else if (channel_id === 0x05 && channel_type === 0xe1) {
>             decoded.water_conv = readUInt16LE(bytes.slice(i, i + 2)) / 10;
>             decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;
>             decoded.water = readFloatLE(bytes.slice(i + 4, i + 8));
>             decoded.gpio_type = 1;
>             i += 8;
>         }
> ```
> 
> Adjusted
> 
> ```
>         // GPIO
>         else if (channel_id === 0x05 && channel_type === 0x00) {
>             decoded.gpio = bytes[i];
>             decoded.gpio_type = 1;
>             i += 1;
>         }
>         // PULSE COUNTER
>         else if (channel_id === 0x05 && channel_type === 0xc8) {
>             decoded.pulse = readUInt32LE(bytes.slice(i, i + 4));
>             decoded.gpio_type = 2;
>             i += 4;
>         }
>         // PULSE COUNTER (v1.3+)
>         else if (channel_id === 0x05 && channel_type === 0xe1) {
>             decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;
>             decoded.pulse = readFloatLE(bytes.slice(i + 4, i + 8));
>             decoded.gpio_type = 2;
>             i += 8;
>         }
> ```
> 
> BACnet using the default codec (this should be 1 = GPIO) 0 is invalid. [image]
> 
> BACnet using the adjusted codec
> 
> [image]

> Shouldn't we also be writing 1 or 2 as gpio_type not 0 and 1? So the MSV work on BACnet. (MSVs start at 1 not 0) and also the history mode selection works?
> 
> Current
> 
> ```
>         // GPIO
>         else if (channel_id === 0x05 && channel_type === 0x00) {
>             decoded.gpio = bytes[i];
>             decoded.gpio_type = 0;
>             i += 1;
>         }
>         // PULSE COUNTER
>         else if (channel_id === 0x05 && channel_type === 0xc8) {
>             decoded.pulse = readUInt32LE(bytes.slice(i, i + 4));
>             decoded.gpio_type = 1;
>             i += 4;
>         }
>         // PULSE COUNTER (v1.3+)
>         else if (channel_id === 0x05 && channel_type === 0xe1) {
>             decoded.water_conv = readUInt16LE(bytes.slice(i, i + 2)) / 10;
>             decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;
>             decoded.water = readFloatLE(bytes.slice(i + 4, i + 8));
>             decoded.gpio_type = 1;
>             i += 8;
>         }
> ```
> 
> Adjusted
> 
> ```
>         // GPIO
>         else if (channel_id === 0x05 && channel_type === 0x00) {
>             decoded.gpio = bytes[i];
>             decoded.gpio_type = 1;
>             i += 1;
>         }
>         // PULSE COUNTER
>         else if (channel_id === 0x05 && channel_type === 0xc8) {
>             decoded.pulse = readUInt32LE(bytes.slice(i, i + 4));
>             decoded.gpio_type = 2;
>             i += 4;
>         }
>         // PULSE COUNTER (v1.3+)
>         else if (channel_id === 0x05 && channel_type === 0xe1) {
>             decoded.pulse_conv = readUInt16LE(bytes.slice(i + 2, i + 4)) / 10;
>             decoded.pulse = readFloatLE(bytes.slice(i + 4, i + 8));
>             decoded.gpio_type = 2;
>             i += 8;
>         }
> ```
> 
> BACnet using the default codec (this should be 1 = GPIO) 0 is invalid. [image]
> 
> BACnet using the adjusted codec
> 
> [image]

The gpio_type of the Milesight EM300-DI is either counter or digital, meaning it has only two states. You can try defining the bacnet_type of gpio_type as "binary_input_object".

Charles-Milesight, 2025-04-12:
> Also in history v2, if we are setting mode/gpio_type to 0 and 1, how do we ever get the mode === 2 to be true.
> 
> Current
> 
> ```
>             if (mode === 1) {
>                 data.gpio = bytes[i + 9];
>             } else if (mode === 2) {
>                 data.water_conv = readUInt16LE(bytes.slice(i + 10, i + 12)) / 10;
>                 data.pulse_conv = readUInt16LE(bytes.slice(i + 12, i + 14)) / 10;
>                 data.water = readFloatLE(bytes.slice(i + 14, i + 18));
>             }
> ```
> 
> Adjusted (unsure if this is accurate, but pulse is missing)
> 
> ```
>             if (mode === 1) {
>                 data.gpio = bytes[i + 9];
>             } else if (mode === 2) {
>                 data.water_conv = readUInt16LE(bytes.slice(i + 10, i + 12)) / 10;
>                 data.pulse_conv = readUInt16LE(bytes.slice(i + 12, i + 14)) / 10;
>                 data.water = readFloatLE(bytes.slice(i + 14, i + 18));
>                 data.pulse = readFloatLE(bytes.slice(i + 14, i + 18));
>             }
> ```

> Also in history v2, if we are setting mode/gpio_type to 0 and 1, how do we ever get the mode === 2 to be true.
> 
> Current
> 
> ```
>             if (mode === 1) {
>                 data.gpio = bytes[i + 9];
>             } else if (mode === 2) {
>                 data.water_conv = readUInt16LE(bytes.slice(i + 10, i + 12)) / 10;
>                 data.pulse_conv = readUInt16LE(bytes.slice(i + 12, i + 14)) / 10;
>                 data.water = readFloatLE(bytes.slice(i + 14, i + 18));
>             }
> ```
> 
> Adjusted (unsure if this is accurate, but pulse is missing)
> 
> ```
>             if (mode === 1) {
>                 data.gpio = bytes[i + 9];
>             } else if (mode === 2) {
>                 data.water_conv = readUInt16LE(bytes.slice(i + 10, i + 12)) / 10;
>                 data.pulse_conv = readUInt16LE(bytes.slice(i + 12, i + 14)) / 10;
>                 data.water = readFloatLE(bytes.slice(i + 14, i + 18));
>                 data.pulse = readFloatLE(bytes.slice(i + 14, i + 18));
>             }
> ```

In the historical data of the EM300-DI, the "mode" represents the interface type. If the mode is 1, it indicates that the interface type is digital; if the mode is 2, it indicates that the interface type is counter. If you cannot see the counter, please ensure that the type of the related BACnet Point is defined as MSV. Additionally, if you are using the built-in NS of the Milesight gateway, please ensure that the Payload_Codec of this EM300-DI has an Object Mapping Function defined for this BACnet Point. For the method of defining the Object Mapping Function, you can refer to this article link: https://support.milesight-iot.com/support/solutions/articles/73000600064-how-to-use-payload-codec-on-milesight-gateway