Milesight-IoT/SensorDecoders #71: vs13x_encoder.js

tof-arch, 2025-04-15:
I found two problems due to the same construction, so most certainly coming from the same function. In the first case, I try to activate the wifi, starting from the example provides, in the second, I try to reboot, always starting from the example provides.

//test
console.log('codage de l\'activation du wifi:\n');
console.log(milesightDeviceEncode({ "wifi_enable": 1 }));

-> result

[image]

//test

console.log('codage du reboot:\n');
console.log(milesightDeviceEncode({ "reboot": 1 }));

[image]

milesight-sway, 2025-04-15:
According `RAW_VALUE=0x00` 
https://github.com/Milesight-IoT/SensorDecoders/blob/e53c6fc317d2b0351b8ffaec315a733c8a1b63b5/VS_Series/VS13x/VS13x_Encoder.js#L8

in RAW_VALUE=0x00;
example data: 
- `{ "reboot": 'yes" }`, means reboot
- `{ "wifi_enable": "enable" }`, means wifi enable

in RAW_VALUE=0x01;
example data: 
- `{ "reboot": 1 }`, means reboot
- `{ "wifi_enable": 1 }`, means wifi enable

tof-arch, 2025-04-15:
Ok, I have just tested, the examples are false: line 60, it is necessary to replace { "reboot": 1 } by { "reboot": "yes" }, and line 132, it is necessary to replace { "wifi_enable": 1 } by { "wifi_enable": "enable" }. Possible that there is other example of false, I have not tested the other examples.

milesight-sway, 2025-04-21:
Example

closed 2025-04-21 by milesight-sway