Milesight-IoT/SensorDecoders #84: FT101_Decoder.js does not pass Lat/Long in payload

techfixpros, 2025-07-17:
FT101
LoRaWAN Version 1.0.3
Work Mode Class A
Join Type OTAA
App Version 1.0.4
Firmware Version 1.3
Hardware Version 1.0

FT101 is onboarded and communicating with LNS, and codec is properly decoding. Values passed in payload are: 

Status, FW Version HW Version, IPSO Version, Class, Serial, TSL Version, RSSI, SF, SNR, and Tx Power.

Latitude and Longitude are displayed in FT101 app while running Signal Evaluation, Real-Time Testing, and Noise Scan, and are included in the exported reports. Inclusion of Lat/Long in the payload will allow 3rd party integration with application server and provide mapping abilities outside of the current exported report method. The decoder already has a function detailing that Lat/Long should/could be decoded from the uplink.

`        }
        // LOCATION
        else if (channel_id === 0x03 && channel_type === 0xa1) {
            decoded.longitude = readInt32LE(bytes.slice(i, i + 4)) / 1000000;
            decoded.latitude = readInt32LE(bytes.slice(i + 4, i + 8)) / 1000000;
            i += 8;
        }
`

FengFuLiu, 2025-07-18:
Starting from version **v1.0.3**, the Field Tester app automatically reports test results when you stop the Real-Time Testing feature. These results should include latitude and longitude values (with up to 6 decimal places). I’ve just tested this on my end, and the reported data does include the coordinates. You can try running the feature to verify.

[image]

techfixpros, 2025-07-29:
I will try to try it again this week. The tester is on 1.0.3, and I downloaded a fresh copy and upgraded again in case I had a bugged version. Ran numerous tests in real-time last week, and prior-to deleted the latitude and longitude from chirpstack measurement page to see if the two data points were discovered. They were not.

I may delete the device and re-onboard it, and test again. The exported .csv do have the correct lat/long.

techfixpros, 2025-07-30:
It did emit the objects, but although the app shows the Lat/Long correctly, the payload only shows 0 for the values.

`    "data": "A6EAAAAAAAAAAASiR/0jAAWjCQakuAs=",
    "object": {
        "snr": 3.5,
        "sf": 9,
        "longitude": 0,
        "rssi": -69.7,
        "latitude": 0,
        "tx_power": 30
    },`

techfixpros, 2025-12-15:
My apologies, was not monitoring this. Will try this week and report back, thanks.